CelliesProjects/ESP32_VS1053_Stream #3: Compiler errors

ws2940, 2021-08-02:
Hi Cellie,
thank you very much for providing this piece of software! Looking forward to make use of it.
Compiling the simple example, I get these messages:

```
Arduino: 1.8.15 (Linux), Board: "DOIT ESP32 DEVKIT V1, 80MHz, 921600, None"

/home/wolfgang/Arduino/libraries/ESP32_VS1053_Stream/src/ESP32_VS1053_Stream.cpp: In member function 'bool ESP32_VS1053_Stream::startDecoder(uint8_t, uint8_t, uint8_t)':
/home/wolfgang/Arduino/libraries/ESP32_VS1053_Stream/src/ESP32_VS1053_Stream.cpp:60:14: error: 'class VS1053' has no member named 'loadDefaultVs1053Patches'
     _vs1053->loadDefaultVs1053Patches();
              ^
/home/wolfgang/Arduino/libraries/ESP32_VS1053_Stream/src/ESP32_VS1053_Stream.cpp: In member function 'bool ESP32_VS1053_Stream::connecttohost(const String&)':
/home/wolfgang/Arduino/libraries/ESP32_VS1053_Stream/src/ESP32_VS1053_Stream.cpp:126:12: error: 'class HTTPClient' has no member named 'setFollowRedirects'
     _http->setFollowRedirects(HTTPC_STRICT_FOLLOW_REDIRECTS);
            ^
/home/wolfgang/Arduino/libraries/ESP32_VS1053_Stream/src/ESP32_VS1053_Stream.cpp:126:31: error: 'HTTPC_STRICT_FOLLOW_REDIRECTS' was not declared in this scope
     _http->setFollowRedirects(HTTPC_STRICT_FOLLOW_REDIRECTS);
                               ^
exit status 1
Error compiling for board DOIT ESP32 DEVKIT V1.

```
Could you maybe point me to how to resolve that ? 

Thank you very much & kind regards,
Wolfgang

ws2940, 2021-08-03:
Now I found this issue is one of the ESP_VS1053_Library, not of the ESP32_VS1053_Stream library. Obviously your code refers to a newer version than the latest official one.

CelliesProjects, 2021-08-04:
Yes, that could be your issue.
If you clone the master branch from `ESP_VS1053_Library` then it should work. 

In a linux terminal you can use `git clone https://github.com/baldram/ESP_VS1053_Library` (in the Arduino libraries folder) to get the master branch.

CelliesProjects, 2021-08-07:
@ws2940 
Version 1.1.4 of `baldram/ESP_VS1053_Library` is released. 
That release should be compatible with  `ESP32_VS1053_Stream`.

Could you please check and report if this solves your issue?

ws2940, 2021-08-08:
Thank you very much for notifying me! With this version everything compliles smoothly. :-)

closed 2021-08-09 by CelliesProjects