CelliesProjects/ESP32_VS1053_Stream #6: How to use other DAC chip type NS4168?

franciscogimeno2000, 2022-02-23:
Hello, thanks for your example codes. 
I am using a m5satck ATOM Speaker Kit with a NS4168 chip decoder, could you tell me how to implement the library of this chip so that it works with your example? 

[https://shop.m5stack.com/products/atom-speaker-kit-ns4168?_pos=1&_sid=968a46ad3&_ss=r](url)

Thanks.

CelliesProjects, 2022-02-23:
That would be unpossible as the ns4168 is a dac and not a codec. (So it expects/plays samples and not a coded stream) 

See for example https://www.analogictips.com/what-is-a-codec/

You might try https://github.com/schreibfaul1/ESP32-audioI2S/ to decode the streams. I used that lib with a couple of different dacs and it should be fairly easy to get it running.

The stream has to be decoded before you send it to the dac which would be done by ESP32audioI2S which sends the decoded data (the samples) to the ns4168.

closed 2022-02-23 by CelliesProjects

franciscogimeno2000, 2022-02-23:
OK thank you very much.

CelliesProjects, 2022-02-23:
Just try the example from https://github.com/schreibfaul1/ESP32-audioI2S/wiki#a-simple-project-to-receive-a-webstream, change the pin numbers to the numbers used by the ns4168 and it could very well be that it works.

franciscogimeno2000, 2022-02-24:
ok thanks, i'll try it.

franciscogimeno2000, 2022-02-24:
It seems to me that it is better to use a codec chip than a DAC. Faster, better quality and less problems.

CelliesProjects, 2022-02-24:
I can only agree with that. I started out with software decoding and after a lot of hassling ended up -very satisfied- with a hardware decoder.

franciscogimeno2000, 2022-02-24:
I have been using your library and with the codec chip vs1053 the sound is perfect.
One question, is your library capable of automatically reconnecting if the server goes down? Or do I have to read the playback status and launch it again

franciscogimeno2000, 2022-02-24:
Another thing, if while it is playing I deactivate the server, the ESP32 gives a panic message and restarts. 
I don't know if it happened to you.

Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled. Core 1 register dump:
PC      : 0x400d3e74  PS      : 0x00060430  A0      : 0x800d3f3c  A1      : 0x3ffb1e40  
A2      : 0x00000000  A3      : 0x3ffb1e8f  A4      : 0x00000001  A5      : 0x00000001  
A6      : 0x0000059c  A7      : 0x00000000  A8      : 0x00000000  A9      : 0x00000000  
A10     : 0x00000000  A11     : 0x3ffcbc00  A12     : 0x00000050  A13     : 0x00000001  
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x0000000a  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000008  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  

ELF file SHA256: 0000000000000000

Backtrace: 0x400d3e74:0x3ffb1e40 0x400d3f39:0x3ffb1e60 0x401623d2:0x3ffb1e80 0x400d66a9:0x3ffb1eb0 0x400d66d5:0x3ffb1ed0 0x40162371:0x3ffb1ef0 0x400d136a:0x3ffb1f10 0x400d1f7e:0x3ffb1f50 0x400d0966:0x3ffb1f90 0x400d7764:0x3ffb1fb0 0x4008990e:0x3ffb1fd0

Rebooting...

CelliesProjects, 2022-02-24:
Interesting. I assumed that radio streams never stop. 
And as you probably know assumption is the mother of all fuckups so there you go.

What setup do you use to start/stop a stream? I want to test this scenario in my network.

franciscogimeno2000, 2022-02-24:
Use any vlc type player and generate a streaming of an mp3 and when it is playing do not hit the vlc stop, close it directly.

franciscogimeno2000, 2022-02-24:
I have also tried to remove the wifi while playing, and in that scenario no error occurs.
With which I believe that the problem is in the reproduction, not in the management of the wifi or the client.

CelliesProjects, 2022-02-24:
Thanks, got the streaming part running. Next thing is debugging. 
Stay tuned...

franciscogimeno2000, 2022-02-24:
ok

franciscogimeno2000, 2022-02-25:
One question... do you know how to synchronize several esp32 so that when connected they play the audio at the same time?

I have read that the WMA protocol sends the real time of the current position of the player in the encoding. But I don't know how to read this.

CelliesProjects, 2022-02-28:
I am getting `"application/octet-stream"` headers from the local vlc streamer on all filetypes. You are using macOS of windows?

franciscogimeno2000, 2022-02-28:
win