CelliesProjects/ESP32_VS1053_Stream #7: codec: UNKNOWN library not start any stream

Sentinel8000, 2022-05-19:
I got this messages from serial monitor:

0:18:00.977 -> Simple vs1053 Streaming example.
20:18:03.259 -> wifi connected - starting decoder
20:18:06.519 -> decoder running - starting stream
20:18:06.566 -> codec: UNKNOWN

I tryed several url:

http://streams.regenbogen.de/rr-eurodance-128-mp3

http://mp3.ffh.de/ffhchannels/hqeurodance.mp3

http://mp3.hitradiort1.c.nmdn.net/rt1eurodancewl/livestream.mp3

http://89.39.189.52:8000/stream/1/

I have used expressif esp32 board with vs1053, set these pins:

#define VS1053_CS     5
#define VS1053_DCS    16
#define VS1053_DREQ   4

CelliesProjects, 2022-05-20:
Hmm not really sure. All the stream urls above work ok for me.
I think you are trying the example with only changed pins?

CelliesProjects, 2022-05-20:
I just compiled with esp32 core version 2.0.3 and then I can reproduce your problem.

With 2.0.2 it works for me. Could you please try to compile and flash with 2.0.2?

Sentinel8000, 2022-05-21:
You have right, thats was the problem, now working well:) You need notice this in description to prevent this issue or check what is the problem with the new core. 

Thank you for investigation, have a nice day.

Sentinel8000, 2022-05-21:
Just a question - this is not issue, because im not pro developer, how can i display on serial monitor just one function, not the full function loop? example if i want see just the audio_showstreamtitle?

CelliesProjects, 2022-05-21:
Thanks for checking out the problem. Something has changed between 2.0.2 and 2.0.3.

For now just use 2.0.2, until I debug this issue. 

About `audio_showstreamtitle`, `audio_showstation` and `audio_eof_stream`:

You can use these routines, but you don't have to.

Sentinel8000, 2022-05-21:
I try simple make this, using your included example, but im unable to compile. 

I dont know what a parameter i need give for audioshowstreamtitle:

```

#include <VS1053.h>            
#include <ESP32_VS1053_Stream.h>

#define VS1053_CS     5
#define VS1053_DCS    16
#define VS1053_DREQ   4

ESP32_VS1053_Stream stream;

const char* SSID = "xxxx";
const char* PSK = "xxxx";

void setup() {
    Serial.begin(115200);
    WiFi.begin(SSID, PSK);

    while (!WiFi.isConnected())
        delay(10);

    SPI.begin();  
    stream.startDecoder(VS1053_CS, VS1053_DCS, VS1053_DREQ);
    stream.connecttohost("http://mp3.ffh.de/ffhchannels/hqeurodance.mp3");

}

void loop() {

   audio_showstreamtitle();   //?

}

```
Possible to run in this form and/or impossible or i have forget something?

CelliesProjects, 2022-05-21:
The three `audio_xxx` functions are called automatically when stream data becomes available. (And you have an instance of that particular function) 

The data becomes available through `const char* info`.
If you want to use that data later on in your program you will have to save it. 
In the example below the station name is saved as `String currentStation` and can be used throughout the rest of the code.

And to proccess the stream `stream.loop()` has to be called every couple of milliseconds. 

```c++
#include <VS1053.h>            
#include <ESP32_VS1053_Stream.h>

#define VS1053_CS     5
#define VS1053_DCS    16
#define VS1053_DREQ   4

ESP32_VS1053_Stream stream;

const char* SSID = "xxxx";
const char* PSK = "xxxx";

String currentSong = "";

void setup() {
    Serial.begin(115200);
    WiFi.begin(SSID, PSK);

    while (!WiFi.isConnected())
        delay(10);

    SPI.begin();  
    stream.startDecoder(VS1053_CS, VS1053_DCS, VS1053_DREQ);
    stream.connecttohost("http://mp3.ffh.de/ffhchannels/hqeurodance.mp3");

}

void loop() {
    stream.loop();
}

void audio_showstreamtitle(const char* info) {
    currentSong = info;
    Serial.printf("currently playing: %s\n", currentSong.c_str());
}
```

Sentinel8000, 2022-05-21:
This loop call method was for me new, never seen before. 

Translate to my basic language this stream.loop(); call a collection of functions:

audio_showstation(const char* info) 
audio_showstreamtitle(const char* info) 
audio_eof_stream(const char* info) 

 and i can remove what i dont need. 

Also this sream.loop() dont call any other functions what is not part of this library.

I have checked also the sound (ealier i have just want see the decoded url) and i need report, i dont have sound from the vs1053 jack decoder. Please check this too if you have time.

CelliesProjects, 2022-05-22:
It looks like a bug in HTTPClient  -or some underlying code- causes this issue. 
The http.begin() call has some weird behaviour after setting http.setConnectTimeout() as in timing out before the set time has elapsed. 
See https://gitter.im/espressif/arduino-esp32?at=6288f01367db9a59dbdde996

CelliesProjects, 2022-05-22:
And should be fixed 26 days ago in master with espressif/arduino-esp32#6633

Sentinel8000, 2022-05-23:
Ok thanks for the information, i will re check later.

Sentinel8000, 2022-05-28:
If im good understand, after the fix, we can use your library with the new esp32 core too, but we still need wait for the fix ,,official".

Currently i have tested your library working well with esp32 core 2.0.2, i have just a question, and notice/small issue

What if somebody try play a bad url? There is any error handling? Or normaly simple will not start the stream nad i can catch this issue with stream.isRunning() and if i dont receive station name?

And i have noticed durint testing trouble with handling special characters - not typical use case, but can generate weird result if somebody try play not only english streams, example in my case hungarian.  I dont know library can handle UTF8 characters, but currently if showstation or streamtitle have spacial characters like áéűúőöüóí and/or uppercase variants user will receive like this:

TOP FM r⸮di⸮ >> 90s, 00s what looks originaly TOP FM rádió >> 90s, 00s.

I dont know this need hard work to handle or do you want handle this, but i dont opened currenly new issue.

Sentinel8000, 2022-05-28:
also a link for the special character using station:

http://live.topfm.hu:8000/radio.mp3

another one:

http://live.topfm.hu:8000/comedy.mp3

Here showstation have special characters.

CelliesProjects, 2022-05-30:
> What if somebody try play a bad url? There is any error handling? Or normaly simple will not start the stream and i can catch this issue with stream.isRunning() and if i dont receive station name?

connecttohost() has a boolean result you can check. Or use isRunning(). Or currentCodec().

> And i have noticed durint testing trouble with handling special characters - not typical use case, but can generate weird result if somebody try play not only english streams, example in my case [...]

The character encoding upstream can be a real pita.
I 'solved' that issue with https://github.com/CelliesProjects/eStreamPlayer32_VS1053/blob/main/percentEncode.h. 

There are probably better solutions you can find on github.

Sentinel8000, 2022-05-30:
Thank you for the help:) If im add this header + in code:

```
const String urlEncode(const String& s) {
    //https://en.wikipedia.org/wiki/Percent-encoding
    String encodedstr{""};
    for (int i = 0; i < s.length(); i++) {
        switch (s.charAt(i)) {
            case ' ' : encodedstr.concat("%20");
                break;
            case '!' : encodedstr.concat("%21");
                break;
            case '&' : encodedstr.concat("%26");
                break;
            case  39 : encodedstr.concat("%27"); //39 == single quote '
                break;
            default : encodedstr.concat(s.charAt(i));
        }
    }
    ESP_LOGD(TAG, "encoded url: %s", encodedstr.c_str());
    return encodedstr;
}
```

and calling url with so, request encode the url:

`stream.connecttohost(urlEncode(station[currentstation].url));`

im receiving same non encoded stationname:)

But as workaround (by my wifi radio), im just simple adding manual the station name (using Serial monitor input feature) and im display this without using the stationname decoding. The oled driver what im using have special character support and so looking better.

Sentinel8000, 2022-06-02:
i have continued testing my project also the vs1053 stream library features, now with different file types like ogg, aac and special type flac/ogg. Whatever normal ogg stream i try, i get false to streamisrunning and codec: unknow and code fail to play ogg streams, no sound.

Tested normal ogg streams:

http://retro.dancewave.online/retrodance.ogg
https://icast.connectmedia.hu/4765/mr2.ogg
https://www.jazzradio.hu:8000/jazzradio_192.ogg

If i try flac ogg (maybe not supported), library detect codec: ogg and streamisrunning true but no sound.

Flac type ogg stream:

http://retro.dancewave.online/retrodance.flac.ogg

Another question is the vs1053 patches, im not sure, because im not developer, your library load all of these using the standard vs1053?

I just ask this because i saw a nice feature using vs1053 library by one of the updated Edzelf webradio:

[(https://github.com/blotfi/ESP32-Radio-with-Spectrum-analyzer/blob/master/src/Esp32_radio.cpp)]

[(https://github.com/blotfi/ESP32-Radio-with-Spectrum-analyzer/blob/master/src/patches.h)]

Here i just thinked about will be a nice feature use this spectrum analyser ,,patch" but im not sure possible to use with your library/ and how is possible to use.. This patch making possible to somehow read the spectrum bands during decode stream and this can be displayed using example OLED display rectangle draw feature. 

I see included a patches header file and in main code calling this:

```
// spectrumAnalyzerAppl1053b-2.plg
// results in XRAM 0x1810 .. 0x1877
#define BASE 0x1810
void VS1053::getBands()
{
    write_register(SCI_WRAMADDR, BASE+2);
    bands = read_register(SCI_WRAM);
    write_register(SCI_WRAMADDR, BASE+4);
    for (uint8_t i = 0; i < 14; i++) {
        uint8_t val = read_register(SCI_WRAM);
        /* current value in bits 5..0, normally 0..31
        peak value in bits 11..6, normally 0..31 */
        uint8_t cur1 = val & 31;
        uint8_t cur = (cur1*curvol)/30; // big bars
        if ( cur < 0 ) cur = 0;
        else if ( cur > (Spectrum_hy-4) ) cur = Spectrum_hy-4;
        spectrum[i][0] = cur;
    }
}
```

Sentinel8000, 2022-06-18:
Maybe i was blind, in example was not included the bitrate display, but was included in readme section:)

I have tryed somehow make usable the spectrum analyser plugin for vs1053, but with my programing skill almost impossible. because i saw only one example, im not sure this will be working.

Also i have make a mod on library localy, i have found a command in original vs1053 library, what with users possible to set bass/trebble but currently not in using in vs1053 stream:

` if (!_vs1053) {
        ESP_LOGE(TAG, "could not initialize vs1053");
        return false;
    }
    _vs1053->begin();
    _vs1053->loadDefaultVs1053Patches();
    _vs1053->switchToMp3Mode();
    uint8_t rtoneCustom[4] = { 7, 15, 13, 8 } ;	
    _vs1053->setTone(rtoneCustom); //bass treble cutom values 
    setVolume(_volume);
    return true;
}
 `