CelliesProjects/ESP32_VS1053_Stream #8: Unable to play ogg streams, flac ogg playing but no sound

Sentinel8000, 2022-06-05:
When i try play ogg streams with the library i get false to streamisrunning and codec: unknow and code fail to play ogg streams, no sound.  

Tested normal ogg streams:  

http://retro.dancewave.online/retrodance.ogg https://icast.connectmedia.hu/4765/mr2.ogg https://www.jazzradio.hu:8000/jazzradio_192.ogg  
https://icast.connectmedia.hu/4765/mr2.ogg
http://dancewave.online/dance.ogg 

If i try flac ogg (maybe not supported), library detect codec: ogg and streamisrunning true but no sound.  

Tested Flac type ogg stream:  http://retro.dancewave.online/retrodance.flac.ogg

Im not sure, the problem is with the library or the ogg format not supported.

CelliesProjects, 2022-06-06:
An Ogg stream can be two things: 

-  [Ogg](https://en.wikipedia.org/wiki/Ogg): a container format for multimedia files.
The Ogg container can contain several stream formats including FLAC.

-  [Ogg-Vorbis](https://en.wikipedia.org/wiki/Ogg_Vorbis): a codec similar to mp3.

This library only plays files and streams encoded with the Ogg-Vorbis codec. So no FLAC encoded files.

Any Ogg files containing streams that are not mp3, aac, aac+ or Ogg-Vorbis will not be recognised or played.

CelliesProjects, 2022-06-06:
See https://www.vlsi.fi/en/support/software/vs10xxpatches.html

CelliesProjects, 2022-06-08:
Just pushed 99eec8a which should fix at least the first 4 streams.

commit 99eec8a3 (2022-06-08): This commit should partially fix #8.
closed 2022-06-08 by CelliesProjects