CelliesProjects/ESP32_VS1053_Stream #9: ESP32 core 2.0.4 most of the stations do not play anymore

3KUdelta, 2022-07-16:
Hello, I am a big fan of your library and I am using it for my web radio. 

This is just to inform that after the update to ESP32 2.0.4 most of my favourite station do not play anymore. They just play perfect on 2.0.2 (so I stay on 2.0.2 for the moment). I am using a ESP32 Dev Board V1 with plain vanilla settings.

Just one station is still running:

`  "https://stream.srg-ssr.ch/m/rsp/mp3_128", `

where as these stations do not play anymore:

```
  "http://mp3.ffh.de/radioffh/hqlivestream.mp3", 
  "http://mp3.ffh.de/ffhchannels/hqlounge.mp3", 
  "http://mp3.ffh.de/ffhchannels/hqsummerfeeling.mp3", 
  "http://mp3.ffh.de/ffhchannels/hqvoting.mp3", 
  "http://deluxe.hoerradar.de/deluxe-easy-mp3-hq",
  "http://ais-sa5.cdnstream1.com/b05055_128mp3",
  "http://energyhits.ice.infomaniak.ch/energyhits-high.mp3",
  "http://energymetime.ice.infomaniak.ch/energymetime-high.mp3" 
```
It would be great if you could keep your lovely library updated to the latest Espressif versions.

3KUdelta, 2022-07-16:
I just realized that this is the only https:// one.

3KUdelta, 2022-07-16:
Setting all to https does the trick and they at least play again. 
But for these stations the sound gets frequently interrupted:
 ```
 "https://ais-sa5.cdnstream1.com/b05055_128mp3",
  "http://deluxe.hoerradar.de/deluxe-easy-mp3-hq",
```

Sentinel8000, 2022-07-17:
You need check the issue list before post issue, i have reported ealier since esp32 core.2.0.3 library have problem, i can confirm just with 2.0.2 working well, 2.03 not re-checked. 

Developer noticed in previous issue the problem comes from the esp32 core itself, also do not forget the library using the base vs1053 library for working, this is a ,,plugin" for vs1053.

Also common rule, never use the newest driver for a board if your other librarys not required. New core drivers can have bugs.

Test with esp32 core 2.0.2 if you have problem with 2.0.4 or with 2.0.3. If you still have problem, you need report.

Im using without problem with core 2.0.2.

3KUdelta, 2022-07-17:
Agree - actually 1.6 runs smoothest. This was just a heads-up.

closed 2022-07-17 by 3KUdelta

CelliesProjects, 2022-07-20:
I am aware of the 2.0.3 and later issues. ATM I am really busy IRL. So for now just compile with 1.0.6 until I find the time to debug and fix these issues.