CelliesProjects/ESP32_VS1053_Stream #10: Stream name not decoded by one station, reason currently unknow

Sentinel8000, 2022-07-29:
Library working well, since weeks using still the esp32 board 2.0.2 core. 

I have noticed trouble only by one station:

https://mediagw.e-tiger.net/stream/coolfm

Somehow library wont decode the station name, just the title -  however wont have special characters: coolfm

CelliesProjects, 2022-08-01:
Some radio stations do not send the station name over the stream. You can always call `audio_showstation()` yourself with the name just before connecting to the host, I also do that in my streamplayer.

CelliesProjects, 2022-08-01:
Closing this for now. Feel free to re-open if needed.

Sentinel8000, 2022-08-01:
wait 1 sec:)

Sentinel8000, 2022-08-01:
Yeah, i know possible not all station sending both information.

I have used winamp to check this, if i good know winamp can read this icy name, if im checking File info, but in this case i see under Station: coolfm.

CelliesProjects, 2022-08-01:
Let me check...

CelliesProjects, 2022-08-01:
The station info is sent a bit strange via icy:
```bash
[ 19415][I][eStreamPlayer32_VS1053.ino:819] handlePastedUrl(): STARTING new url: https://mediagw.e-tiger.net/stream/coolfm with 0 items in playList
[ 19418][I][eStreamPlayer32_VS1053.ino:890] startCurrentItem(): Starting playlist item: 0
[ 19425][W][HTTPClient.cpp:258] beginInternal(): unexpected protocol: https, expected http
[ 19495][I][ssl_client.cpp:173] start_ssl_client(): WARNING: Skipping SSL Verification. INSECURE!
[ 21330][I][ESP32_VS1053_Stream.cpp:234] _parseMetaData(): metadata: StreamTitle='Joel Corry / David Guetta / Bryson Tiller - What Would You Do';
[ 22317][I][ESP32_VS1053_Stream.cpp:234] _parseMetaData(): metadata: StreamTitle='COOL FM - no.1 hit music - coolfm.hu';
[ 30310][I][ESP32_VS1053_Stream.cpp:234] _parseMetaData(): metadata: StreamTitle='Maroon 5 / SZA - What Lovers Do';
```
The info is in the `StreamTitle` data.

Sentinel8000, 2022-08-01:
Ah, this lookslike a icy server issue, not your libraray issue.

Sentinel8000, 2022-08-01:
Im not a developer, but, if you can wait few sec, i give you a youtube link, you can check yourself how i have used your libraray.

Sentinel8000, 2022-08-01:
https://youtu.be/BxsbCReP_Tk

CelliesProjects, 2022-08-01:
Cool! Nice project!

>> On my webradio, this generate a weird effect, im store in a variable the station name, and if im switch to a another stream, because the function not read / push any new station name, i see the previous station name:) with the current artist/track name.

Also, send the station name just before connecting so you have the correct name on the display.

CelliesProjects, 2022-08-01:
Also, the url above also works over http.

Sentinel8000, 2022-08-01:
I have checked whats display the serial monitor under arduino ide, this problem comes because the variable wont get any new value as station name, as workaround i will try reset this variable after every change.

The current problem comes from, the function itself a trigger for variable, if i get new station name, knocks out the previous station name.

In first version i have a more primitive solution: i have stored a manual inputted Station names, not using the station name function.

Sentinel8000, 2022-08-01:
I have a another nice feature whats is not in video. The esp32 runs a webserver itself, and you can enter the statiuon urls via webbrowser.

CelliesProjects, 2022-08-01:
Ha! beat you to it.

Check out

https://github.com/CelliesProjects/eStreamPlayer32_VS1053

With websocket push connection for streamtitle and station name.

Sentinel8000, 2022-08-01:
Yeah, i have checked your project, you have maked a local music player what wont need any oled display. Yours more adanced like my:) I just maked a simple web management site, without any extra controll.

With oled i have lot troubles, driver etc.

Sentinel8000, 2022-08-01:
And i have maked a little mod in your library to give bass boost:

void ESP32_VS1053_Stream::setBassOn() {
	uint8_t rtoneCustom[4] = { 7, 15, 13, 8 } ;	
	_vs1053->setTone(rtoneCustom); //bass treble
}

void ESP32_VS1053_Stream::setBassOff() {
	uint8_t rtoneCustom[4] = { 0, 15, 0, 8 } ;	
	_vs1053->setTone(rtoneCustom); //bass treble
}

Sentinel8000, 2022-08-02:
Just a short follow up: 

I have contacted with the coolfm support, the team confirmed the problem and waiting for developers to fix.

Also thank you for you for the investigation.