CelliesProjects/ESP32_VS1053_Stream #12: ICY name problem?

Sentinel8000, 2022-10-21:
Im using more weeks my wifi radio what using your library. We know some station have configuration problem, and impossible to receive the station name. 

But there was two station, whats worked well, but now, i dont know why the library or my code unable to decode the icy name.

I have asked the radio server owner about any changes in these weeks, but response was, there is no problem icy name is ok. He sendet me about a picture, what i have attached.

Can you check with your method this problem?

http://retro.dancewave.online/retrodance.ogg

http://retro.dancewave.online/retrodance.mp3

Sentinel8000, 2022-10-21:
[image]

CelliesProjects, 2022-10-21:
That screenshot is from a different url than the 2 you posted above. And the posted links give no station name back.

http://dancewave.online/dance.ogg gives:
```
$ curl -I -X GET -L http://dancewave.online/dance.ogg
HTTP/1.1 302 Moved Temporarily
Server: nginx
Date: Fri, 21 Oct 2022 14:23:34 GMT
Content-Type: text/html
Content-Length: 0
Connection: keep-alive
Location: http://stream.dancewave.online:8080/dance.ogg
Access-Control-Allow-Origin: *
Access-Control-Allow-Headers: *
Access-Control-Allow-Credentials: true

HTTP/1.0 200 OK
Content-Type: application/ogg
Date: Fri, 21 Oct 2022 14:23:34 GMT
icy-br:160
ice-audio-info: channels=2;bitrate=160;quality=5;samplerate=44100
icy-br:160
icy-description:All about Dance from 2000 till today!
icy-genre:Club Dance Electronic House Trance
icy-name:Dance Wave!
icy-pub:1
icy-sr:44100
icy-url:https://dancewave.online
icy-vbr:1
Server: Icecast 2.4.0-kh15 (MSCP)
Cache-Control: no-cache, no-store
Expires: Mon, 26 Jul 1997 05:00:00 GMT
Connection: Close
Access-Control-Allow-Origin: *
Access-Control-Allow-Headers: Origin, Accept, X-Requested-With, Content-Type, Icy-MetaData
Access-Control-Allow-Methods: GET, OPTIONS, SOURCE, PUT, HEAD, STATS
```
There is a `icy-name` header but that is a different url from the ones you posted above.

And http://retro.dancewave.online/retrodance.ogg gives:
```
$ curl -I -X GET -L http://retro.dancewave.online/retrodance.ogg
HTTP/1.1 302 Moved Temporarily
Server: nginx
Date: Fri, 21 Oct 2022 14:26:14 GMT
Content-Type: text/html
Content-Length: 0
Connection: keep-alive
Location: http://stream2.dancewave.online:8080/retrodance.ogg
Access-Control-Allow-Origin: *
Access-Control-Allow-Headers: *
Access-Control-Allow-Credentials: true

HTTP/1.1 200 OK
Date: Fri, 21 Oct 2022 14:26:14 GMT
Content-Type: application/ogg
Transfer-Encoding: chunked
Connection: keep-alive
Cache-Control: no-cache, no-store
Expires: Mon, 26 Jul 1997 05:00:00 GMT
Access-Control-Allow-Origin: *
Access-Control-Allow-Headers: Origin, Accept, X-Requested-With, Content-Type, Icy-MetaData
Access-Control-Allow-Methods: GET, OPTIONS, SOURCE, PUT, HEAD, STATS
CF-Cache-Status: BYPASS
Report-To: {"endpoints":[{"url":"https:\/\/a.nel.cloudflare.com\/report\/v3?s=DAc0%2FAV5dPRucCHWm4Ryx26nrXOntZNavLTZ60dCg3IEodCP9uR70QBUTO%2FPJoA3Gcm%2FBM6vmu4pZrEixx2LllZjmv8iPlf2Lvy%2FVFYxNJoYTQxhBphxPv3bhMf0r0kapBeko7LJfsJW3CjP4a4FVw%3D%3D"}],"group":"cf-nel","max_age":604800}
NEL: {"success_fraction":0,"report_to":"cf-nel","max_age":604800}
Server: cloudflare
CF-RAY: 75daaac8ca4eb89c-AMS
alt-svc: h3=":443"; ma=86400, h3-29=":443"; ma=86400
```
So there is no name in the http://retro.dancewave.online/retrodance.ogg station.

I will close this for now as there is no real issue except confusion about urls.

It seems the streaming service just dropped the `icy-name` header.

Feel free to reopen if there is a real issue.

closed 2022-10-21 by CelliesProjects

Sentinel8000, 2022-10-21:
Sorry, was too late, i have not watched, the radio server support gived me the secondary channel url and tested this.

So, you have right on the used url, s there is no metadata/icy data, not my wifi radio/your library has failed.

thank you for the investigation, next i will check myself with these commands before report anything.

Now im waiting for the server side answer, why and what has changed since.