CelliesProjects/ESP32_VS1053_Stream #22: How to use an existing buffer

lidense, 2023-12-13:
Hi, I have used your app to listen to some radios and works great. Now I have developed a small app that provides a buffer in a loop with some aac audio data in chunks of about 1600 bytes coming from an decoded HLS stream. The buffer has this format:  uint8 * data.
I can play the data using Adafruit VS1053 library with this code 
  if(vs_player.readyForData())
  vs_player.playData(data,read);
but the sound is hashed and sometimes some bits are skipped.
I thought that using a circular buffer like esp32 Stream vs1053 might be better but I don't see how to feed the data to the stream.
I see that the main function stream.connectTohost is used to create a stream from a website, how can I instead feed by data buffer to the stream and get it played from vs1053?
Thanks for your help.

CelliesProjects, 2023-12-14:
Hi lidense.

> how can I instead feed by data buffer to the stream and get it played from vs1053?

Not that easy I think. This library is intended purely as a (internet) stream client. 
All internal functions depend on a `WiFiClient` pointer to function. 
Something like you suggest would require a separate codepath in the loop to function.
And probably several helper functions.

CelliesProjects, 2023-12-14:
If that is all your app is doing then you could maybe use just the VS1053 library and use the 
VS1053 [`playChunk()`](https://github.com/baldram/ESP_VS1053_Library/blob/master/src/VS1053.h#L136-L137) function.

It says that the function blocks, but that is only true if you dont use [`data_request()`](https://github.com/baldram/ESP_VS1053_Library/blob/master/src/VS1053.h#L166-L167) as a good to go signal.

This library uses `playChunk()` exactly in that way. 

While `data_request()` is true, send 32 bytes until data is gone.

https://github.com/CelliesProjects/ESP32_VS1053_Stream/blob/master/src/ESP32_VS1053_Stream.cpp#L584-L596

If you dont mind blocking use it to send the whole buffer:

`void playChunk(uint8_t *data, size_t len);`

lidense, 2023-12-15:
OK thanks for your answer, I will try.
Regards
Dave

----- Mail original -----

https://github.com/CelliesProjects/ESP32_VS1053_Stream/blob/master/src/ESP32_VS1053_Stream.cpp#L584-L596

Frank-Bemelman, 2025-06-30:
Works like a charm. I use it to play that famous windows XP startup tune, when my radio boots. Converted a small mp3 file with that sound to an array, stored in PROGMEM. Just 2 lines at the bottom of setup().

```
// windows tune
VS1053 player(VS1053_CS, VS1053_DCS, VS1053_DREQ);
player.playChunk((uint8_t *)mp3_winxpstart, sizeof(mp3_winxpstart));

```

Very happy with both these libraries.
Thank you so much!