CelliesProjects/ESP32_VS1053_Stream #23: 'class VS1053' has no member named 'getChipVersion'

dieseltravis, 2023-12-30:
I get an error when compiling in Arduino 1.8.19: 
```
D:\Arduino\libraries\ESP32_VS1053_Stream\src\ESP32_VS1053_Stream.cpp: In member function 'bool ESP32_VS1053_Stream::startDecoder(uint8_t, uint8_t, uint8_t)':
D:\Arduino\libraries\ESP32_VS1053_Stream\src\ESP32_VS1053_Stream.cpp:153:18: error: 'class VS1053' has no member named 'getChipVersion'
     if (_vs1053->getChipVersion() == 4)
                  ^~~~~~~~~~~~~~
```

I installed latest version of VS1053 (1.1.4) is there an earlier version that should be used? I can see it in the source:
https://github.com/baldram/ESP_VS1053_Library/blob/824f82cff459bba2b3184b29bc21f7547bc6c516/src/VS1053.cpp#L378C23-L378C23

CelliesProjects, 2023-12-31:
@dieseltravis You are right.  Get chip version was only added in https://github.com/baldram/ESP_VS1053_Library/commit/ba1803f75722a36f3e9f539129e885bea3c60f71 after the 1.1.4 release

See https://github.com/baldram/ESP_VS1053_Library/compare/1.1.4...master

Pull the master branch, recompile and it will probably work.

dieseltravis, 2024-01-01:
@CelliesProjects that fixed it, thanks!

closed 2024-01-01 by dieseltravis