CelliesProjects/ESP32_VS1053_Stream #29: Trying to use your basic Example sketch

hausofpayne17, 2024-04-28:
I've added my SSID and psk to the file.

When compiling get:

C:\Documents\Arduino\libraries\ESP32_VS1053_Stream-master\src/ESP32_VS1053_Stream.h:75:5: error: 'StaticRingbuffer_t' does not name a type
     StaticRingbuffer_t *_buffer_struct;

Please help me resolve this issue so I can use your library in my esp32 internet radio project.

Thank you for your creation and support of this library.

Gord

CelliesProjects, 2024-04-28:
What version of the esp32 environment are you using? Arduino IDE or PIO? Need more info.

 Just found https://github.com/CelliesProjects/ESP32_VS1053_Stream/issues/25 that seems about the same problem.

hausofpayne17, 2024-04-29:
Thank you for responding.

The problem was I had REALLY old esp32 board definitions versions on my
device.

Apologies for the inconvenience.

Your demo radio program is working GREAT!

Thank you so much for your work for the community!

Best wishes,

Gord
(Ontario Canada)

CelliesProjects, 2024-04-29:
Closing this issue as resolved.
Pinning this issue as it seems to re-occur.