CelliesProjects/ESP32_VS1053_Stream #31: On https chunked stream, stream->available() returns 31 or less and hangs , but stream unblocks if we read these bytes

vladkozlov69, 2024-05-08:
On some https streams I observe weird behaviour (most probably related to the media server) :

It starts to play, plays less than second and hangs.

[Here](https://github.com/CelliesProjects/ESP32_VS1053_Stream/blob/ba57061cac19f2e2c7c8f7fd300e3702a00c40ed/src/ESP32_VS1053_Stream.cpp#L583) in 

```
void ESP32_VS1053_Stream::_handleChunkedStream(WiFiClient *const stream)
```

call of "stream->available()" returns 31 or less. It's quite strange that it remains the same on subsequent calls until we try to read some bytes from the stream.

So I locally made this change, but not sure whether it is a proper fix or there could be a more elegant way:

```
            if (stream->available() < BYTES_TO_READ)
                break;
```

replaced with 

```
            int BYTES_IN_BUFFER = 0;
            if (stream->available() < BYTES_TO_READ)
            {
                log_i("Not enough data in web stream buffer, trying to read byte-by-byte");
                while(stream->available() && BYTES_IN_BUFFER < VS1053_PLAYBUFFER_SIZE)
                {
                    _vs1053Buffer[BYTES_IN_BUFFER++] = stream->read();
                }
            }
            else
            {
                BYTES_IN_BUFFER = stream->readBytes(_vs1053Buffer, BYTES_TO_READ);
            }
```

Somehow it unblocks/unlocks the stream and after that it runs fine.
I believe there could be more beautiful approach but this rough fix did the trick.

CelliesProjects, 2024-05-08:
@vladkozlov69 Thanks for reporting this. 
Could you send me some of those urls?

vladkozlov69, 2024-05-08:
@CelliesProjects 
    "https://eu3.fastcast4u.com/proxy/twr0"
    "https://stream.purewave.ru:8443/live128.mp3"
    "https://nlradio.stream:8000/nlradio-low.aac"
    
  I use ESP32-WROOM so no ringbuffer there. Maybe with ringbuffer it will work without this fix, but on ESP32-WROOM it plays 500-700 milliseconds and stops because of "underbuffered" stream. stream->available() returns less than 32 until I forcibly read all buffer contents.

CelliesProjects, 2024-05-08:
@vladkozlov69 Could you please test the linked PR?

CelliesProjects, 2024-05-09:
@vladkozlov69 The issue should be solved with the latest commit. 

I tested the three urls above and they run fine. 

Let me know if this can be merged.

vladkozlov69, 2024-05-09:
Hi @CelliesProjects I tested it and it works fine now. Thanks a lot! It can be merged now.

PR #32 (2024-05-09): Fix stream available bug.
This PR should fix #31