CelliesProjects/ESP32_VS1053_Stream #34: WifiClient noy declared

Sikkens99, 2024-06-23:
Since two days I suddenly get this error compiling a program that worked before without problems:
"....libraries\ESP32_VS1053_Stream\src/ESP32_VS1053_Stream.h:78:27: error: 'WiFiClient' has not been declared"
I appreciate your ideas what could be the problem here.

Marten Sikkens

Sikkens99, 2024-06-25:
...and the above-mentioned compilation message is then follwed by "exit status 1".

CelliesProjects, 2024-06-26:
> Since two days I suddenly get this error compiling a program that worked before without problems:

Those are the best errors.

I was toying around with the 3.0.1 esp32 update and there I also got problems with WifiClient not being defined.
Maybe your IDE updated to 3.0.1 without telling you so?

Sikkens99, 2024-06-27:
Indeed I updated the ESP32 IDE to version 3.0.1 without realizing that this could give problems.  I went back to version 2.0.17 and that indeed solved the WiFiClient Not Declared error! Thanks for letiing me know that this could be the problem!

closed 2024-06-27 by Sikkens99