CelliesProjects/ESP32_VS1053_Stream #37: Can I user ESP32 devkit v1

hadi-mansoori, 2024-07-29:
I use this code for ESP32 DEVKIT V1 but can not played the song with a good quality after some seconds will be disconnected .
I must ESP32 S2 or more for live streaming ?

CelliesProjects, 2024-08-01:
The ESP32 DEVKIT V1 should work just fine.
Which stream is giving you issues?

hadi-mansoori, 2024-08-01:
I used ESP32 devkit v1 with this streaming server http://star.jointil.net/proxy/jrn_beat?mp=/stream 
but after some seconds will be disconnected  or you can check this stream http://13.39.128.34:3000/stream
but I test with HTML to play the streaming server it plays completely.

CelliesProjects, 2024-08-01:
It seems the issue was a header thing. Should play fine now.

hadi-mansoori, 2024-08-01:
it played good but after some secound shows me this error .
[  5533][D][ESP32_VS1053_Stream.cpp:727] loop(): startmute is 80 milliseconds
streamtitle: Freq & Liquid Soul - Liquid Frequencies
[ 42453][E][ESP32_VS1053_Stream.cpp:699] loop(): Stream timeout 900 ms
[ 42457][D][HTTPClient.cpp:408] disconnect(): tcp is closed

eof: http://star.jointil.net/proxy/jrn_beat?mp=/stream

CelliesProjects, 2024-08-01:
I just ran the stream again with and without the latest commit applied and now it runs fine both ways.
The stream is running now for 10 minutes and no issues.

```
decoder running - starting playback
showstation: Joint Radio Beat
size: 0
codec: MP3
bitrate: 128kbps
streamtitle: 018 Hux Flux - Stack Babbler (Goa Psy 140 C Mix)
streamtitle: Sensifeel - Olympics
```

commit 96290854 (2024-08-01): Partially fix #37
closed 2024-12-20 by CelliesProjects