CelliesProjects/ESP32_VS1053_Stream #42: Problem playing some Internet radios

ftomasczik, 2024-12-29:
I'm using ESP32:VS1053 and I think it's the best library today to listening Internet radios.

But, I tried to play some station and I get this message after to call to stream.connecttohost() and I can't listen anything: 

Codec: STOPPED
Bitrate: 0 kbps

It's appears on HTTPS and HTTP.

My code is the same as the example for this project.

Here, some examples of radios than don't work:

https://radio2.vip-radios.fm:18009/stream-128kmp3-BossaNovaBrazil
http://listen1.myradio24.com:9000/5967
http://185.33.21.112:11271/
http://strm112.1.fm/bossanova_mobile_mp3

Can anybody help me?

PS: Sorry for my English :(

CelliesProjects, 2024-12-30:
Hi @ftomasczik 
https://radio2.vip-radios.fm:18009/stream-128kmp3-BossaNovaBrazil
gives this:
```bash
~$ curl -I https://radio2.vip-radios.fm:18009/stream-128kmp3-BossaNovaBrazil
HTTP/1.1 502 Bad Gateway
Server: nginx/1.14.2
Date: Mon, 30 Dec 2024 12:55:57 GMT
Content-Type: text/html
Content-Length: 173
Connection: keep-alive
```

Not a stream.

Second stream plays just fine here:
http://listen1.myradio24.com:9000/5967
```bash
simple vs1053 Streaming example
wifi connecting to huiskamer
wifi connected - starting spi bus
spi running - mounting sd card
[  1708][W][sd_diskio.cpp:109] sdWait(): Wait Failed
[  1712][W][sd_diskio.cpp:485] ff_sd_initialize(): sdWait fail ignored, card initialize continues
[  1721][W][sd_diskio.cpp:489] ff_sd_initialize(): GO_IDLE_STATE failed
[  1728][E][sd_diskio.cpp:761] sdcard_mount(): f_mount failed: (3) The physical drive cannot work
[  2236][W][sd_diskio.cpp:109] sdWait(): Wait Failed
[  2240][E][sd_diskio.cpp:128] sdSelectCard(): Select Failed
[  2246][E][main.cpp:17] mountSDcard(): Card Mount Failed
sdcard not found
sdcard mounted -or not-, starting decoder...
decoder running - starting playback
showstation: Auto-DJ
size: 0
codec: MP3
bitrate: 192kbps
streamtitle: Masterboy - Masterboy - Generation Of Love
streamtitle: Masterboy - Masterboy - Generation Of Love
streamtitle: Masterboy - Masterboy - Generation Of Love
```

Third stream also plays fine:
http://185.33.21.112:11271/
```bash
simple vs1053 Streaming example
wifi connecting to huiskamer
[  4997][W][STA.cpp:137] _onStaArduinoEvent(): Reason: 39 - TIMEOUT
[  5003][W][STA.cpp:541] disconnect(): STA already disconnected.
wifi connected - starting spi bus
spi running - mounting sd card
[  5721][W][sd_diskio.cpp:109] sdWait(): Wait Failed
[  5725][W][sd_diskio.cpp:485] ff_sd_initialize(): sdWait fail ignored, card initialize continues
[  5734][W][sd_diskio.cpp:489] ff_sd_initialize(): GO_IDLE_STATE failed
[  5741][E][sd_diskio.cpp:761] sdcard_mount(): f_mount failed: (3) The physical drive cannot work
[  6249][W][sd_diskio.cpp:109] sdWait(): Wait Failed
[  6253][E][sd_diskio.cpp:128] sdSelectCard(): Select Failed
[  6259][E][main.cpp:17] mountSDcard(): Card Mount Failed
sdcard not found
sdcard mounted -or not-, starting decoder...
decoder running - starting playback
showstation: bossanova
size: 0
codec: MP3
bitrate: 192kbps
streamtitle: ZIMBO TRIO - BALANCO ZONA SUL
```

You can check a url with `curl` in linux commandline like this:
```bash
$ curl -I http://listen1.myradio24.com:9000/5967
HTTP/1.0 200 OK
Server: Vmeste v1.5
Date: Mon, 30 Dec 2024 12:56:47 GMT
Expires: Mon, 26 Jul 1997 05:00:00 GMT
Cache-Control: no-store, no-cache
Connection: close
Access-Control-Allow-Origin: *
Access-Control-Allow-Methods: GET, OPTIONS, HEAD
Access-Control-Allow-Headers: Origin, Accept, Content-Type, X-Requested-With
Content-Type: audio/mpeg
icy-br: 192
icy-name: Auto-DJ
icy-djname: Auto-DJ
icy-description: Best Of 90's
icy-url: http://vk.com/90seurodance
icy-pro: 1

```

What you look for is this header:
```bash
Content-Type: audio/mpeg
```
If that is present then the stream will probably play.
Also `application/ogg` or `audio/aac` or `audio/aacp` will probably play.

ftomasczik, 2024-12-31:
Thanks @CelliesProjects for your answer.

Let me explain my new tests for those streams.

Of course, I can play all these stream with VLC or by any browser. 

**a) Case https://radio2.vip-radios.fm:18009/stream-128kmp3-BossaNovaBrazil**

It's true, when I execute curl -I for this url, I get:

```
federico@testing:~$ curl -I https://radio2.vip-radios.fm:18009/stream-128kmp3-BossaNovaBrazil
HTTP/1.1 502 Bad Gateway
Server: nginx/1.14.2
Date: Mon, 30 Dec 2024 23:24:42 GMT
Content-Type: text/html
Content-Length: 173
Connection: keep-alive
```
In my serial port I can see this message and I can't listen anithing (remember, I'm using now the [play-stream.ino](https://github.com/CelliesProjects/ESP32_VS1053_Stream/blob/master/examples/play-stream/play-stream.ino) example) 

```
Station: Vip-Radios.FM - Bossa Nova Brazil
Codec: MP3
Bitrate: 128 kbps
End of stream: https://radio2.vip-radios.fm:18009/stream-128kmp3-BossaNovaBrazil
```

But, if I execute:

```
federico@testing:~$ sudo curl https://radio2.vip-radios.fm:18009/stream-128kmp3-BossaNovaBrazil --output testing.mp3
[sudo] password for federico:
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  159k    0  159k    0     0  26683      0 --:--:--  0:00:06 --:--:-- 34750
```
I can play test.mp3 ok, 

**b) Case http://listen1.myradio24.com:9000/5967**

In this case, my **curl -I** output is:

```
federico@testing:~$ curl -I http://listen1.myradio24.com:9000/5967
HTTP/1.0 200 OK
Server: Vmeste v1.5
Date: Mon, 30 Dec 2024 23:57:06 GMT
Expires: Mon, 26 Jul 1997 05:00:00 GMT
Cache-Control: no-store, no-cache
Connection: close
Access-Control-Allow-Origin: *
Access-Control-Allow-Methods: GET, OPTIONS, HEAD
Access-Control-Allow-Headers: Origin, Accept, Content-Type, X-Requested-With
Content-Type: audio/mpeg
icy-br: 192
icy-name: Auto-DJ
icy-djname: Auto-DJ
icy-description: Best Of 90's
icy-url: http://vk.com/90seurodance
icy-pro: 1
```
So, I should be able to listen this stream. But I get this message by serial port:
```
VS1053 running - starting radio stream
Stream not running - system halted
```
Again, if I try to download the stream:

```
federico@testing:~$ sudo curl http://listen1.myradio24.com:9000/5967 --output test.mp3
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  144k    0  144k    0     0  35082      0 --:--:--  0:00:04 --:--:-- 35078
```
And I can play without any problem my downloaded file **test.mp3**.

**c) Case http://185.33.21.112:11271/**

As you could see, the output for **curl - I** on this url is:

```
federico@testing:~$ curl -I http://185.33.21.112:11271/
HTTP/1.0 200 OK
icy-pub: 1
icy-description: bossanova
icy-url: http://1.fm
icy-name: bossanova
Content-Type: audio/mpeg
icy-metaint: 0
icy-genre: Bossa Nova
Instance-id: 5311f43c7660c783d23233d42ece4efc
Server: AIS Streaming Server 9.1.4
icy-br: 192
Expires: Mon, 26 Jul 1997 05:00:00 GMT
Pragma: no-cache
Cache-Control: no-cache
Connection: close
```

But when try to play this stream I recibe by serial port:
```
VS1053 running - starting radio stream
Stream not running - system halted
```
Well, If I try to save this stream:

```
federico@testing:~$ sudo curl  http://185.33.21.112:11271/ --output test.mp3
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  440k    0  440k    0     0  43169      0 --:--:--  0:00:10 --:--:-- 31889
```

I can play the file test.mp3.

Some many streams are working really well, but I'm trying to understand why those kinds of streams don't work.

Let me know how can I help you.

CelliesProjects, 2024-12-31:
@ftomasczik 
I still can not reproduce your issue here. 
Please run the example on a problematic stream with debug level verbose.
Maybe this will give some more info.

ftomasczik, 2024-12-31:
Thanks @CelliesProjects again.

I have a lot of errors when I try to run the example with debug level verbose (sorry, as you can see, I'm a newbie :)

But running the example with debug level "info" I get the next messages:

**b) Case http://listen1.myradio24.com:9000/5967 and c) Case http://185.33.21.112:11271/**

```
WiFi connected - starting decoder...
VS1053 running - starting radio stream
[  5258][I][NetworkClient.cpp:261] connect(): select returned due to timeout 250 ms for fd 48
[  5267][W][HTTPClient.cpp:1421] returnError(): error(-1): connection refused
[  5274][E][ESP32_VS1053_Stream.cpp:372] connecttohost(): error -1 connection refused
Stream not running - system halted
VS1053 running - starting radio stream
[  4327][I][NetworkClient.cpp:261] connect(): select returned due to timeout 250 ms for fd 48
[  4336][W][HTTPClient.cpp:1421] returnError(): error(-1): connection refused
[  4343][E][ESP32_VS1053_Stream.cpp:372] connecttohost(): error -1 connection refused
Stream not running - system halted
```
I'm using  **arduino-esp32 v3.1.0** and **ESP32_VS1053_Stream v2.1.2**.

Let me know if you need something and Happy New Year from Argentina!

ftomasczik, 2024-12-31:
Sorry, the problems with verbose mode was because in **ESP32_VS1053_Stream.cpp** **bytesToDecoder** and **bytesToRingBuffer** variables are commented.

So, I changed every line:
`
log_d("spend %lu ms stuffing %i bytes in decoder", millis() - START_TIME_MS, bytesToDecoder);`

for:

`log_d("spend %lu ms stuffing %i bytes in decoder", millis() - START_TIME_MS, 0);`

And every line:

`log_d("spend %lu ms stuffing %i bytes in ringbuffer", millis() - START_TIME_MS, bytesToRingBuffer);`

for:

`log_d("spend %lu ms stuffing %i bytes in ringbuffer", millis() - START_TIME_MS, 0);`

After that, I can show verbose logs.

For problematic stream I get this message (I show you only the more important part):
```

[  6075][D][HTTPClient.cpp:293] beginInternal(): protocol: http, host: listen1.myradio24.com port: 9000 url: /5967
[  6086][D][HTTPClient.cpp:574] sendRequest(): request type: 'GET' redirCount: 0

[  6093][D][NetworkManager.cpp:83] hostByName(): Clearing DNS cache
[  6397][D][NetworkManager.cpp:127] hostByName(): DNS found IPv4 136.243.9.89
[  6656][I][NetworkClient.cpp:261] connect(): select returned due to timeout 250 ms for fd 48
[  6664][D][HTTPClient.cpp:1105] connect(): failed connect to listen1.myradio24.com:9000
[  6672][W][HTTPClient.cpp:1421] returnError(): error(-1): connection refused
[  6679][D][ESP32_VS1053_Stream.cpp:253] connecttohost(): Time elapsed during connect: 615
[  6687][E][ESP32_VS1053_Stream.cpp:372] connecttohost(): error -1 connection refused
[  6695][D][HTTPClient.cpp:393] disconnect(): tcp is closed

Stream not running - system halted
```

I hope that can help.

CelliesProjects, 2024-12-31:
@ftomasczik Sorry about the debug statements not compiling.

But in the logs is some info that might help.

>  [  6656][I][NetworkClient.cpp:261] connect(): select returned due to timeout 250 ms for fd 48

and 

>  [  6679][D][ESP32_VS1053_Stream.cpp:253] connecttohost(): Time elapsed during connect: 615

It seems your connection times out.

You could try to set the connection timeouts in `ESP32_VS1053_Stream.h` to some greater value.

These are defined as 
```c++
#define VS1053_CONNECT_TIMEOUT_MS 250
#define VS1053_CONNECT_TIMEOUT_MS_SSL 2500
```

First try something very large like 5000 and see if it helps.

ftomasczik, 2025-01-01:
Oh, it's great!

I modified the value of **VS1053_CONNECT_TIMEOUT_MS** to **5000** and I can now to play any stream!

I trying to find the best value (I think 5000 ms is a lot, I don't now).

Thanks @CelliesProjects!

CelliesProjects, 2025-01-01:
Good to hear! 

I am working on some code to set a custom connection timeout. For now it is a permanent setting, until you set it again to some new value. I think I will make it on a per connection base, not really sure yet. 
So this might not be the final state.

https://github.com/CelliesProjects/ESP32_VS1053_Stream/blob/a69ca61e06f68fec4c44ef804c0c8e7032416ee4/src/ESP32_VS1053_Stream.h#L43

You can test the feature by pulling the master branch.

To use you first set the custom timeout and then you connect.

CelliesProjects, 2025-01-02:
Closing this as it should be fixed with the latest commits.

closed 2025-01-02 by CelliesProjects