CelliesProjects/ESP32_VS1053_Stream #45: Regression with some streams after commit dac5ec9 ?

paulgreg, 2025-01-25:
# Bug Report: Radio France HiFi AAC Streams "Choke" on Master Branch

## Description

Radio France HiFi AAC streams appear to "choke" or fail to play when using the current master branch of the library. This issue affects the following streams:

- http://icecast.radiofrance.fr/fip-hifi.aac
- http://icecast.radiofrance.fr/fipjazz-hifi.aac
- http://icecast.radiofrance.fr/francemusique-hifi.aac

However, these streams work correctly when using commit `dac5ec9`.

## Additional Context

I’ve built a [web radio project](https://hackaday.io/project/194700-esp32-web-radio) using your library, which works great for Radio France streams - thank you for this awesome library!

I also previously built [another project](https://hackaday.io/project/183183-esp32-bluetooth-audio-receiver) to stream audio via Bluetooth, using the same platform (ESP32 + VS1053 chip).

I wanted to merge these two projects into a single device for convenience. To achieve this, I needed to expose the `playChunk` method on the VS1053 component. Therefore, I [forked your project](https://github.com/paulgreg/ESP32_VS1053_Stream_raw) on master branch and added this method.

I had the « choke » problem. I’ve compared both projects and found out that the main change was coming from the library versioon.

I’ve created another branch from the working commit, added my `playChunk` method and my project worked for all streams and also for bluetooth playback. So it seems that a regression was introduced between theses 2 commits.

Thanks again for your hard work on this library—it’s been a fantastic tool for my projects!

CelliesProjects, 2025-01-26:
@paulgreg Thanks for reporting this. I could reproduce your issue and will look into it.

The diff: https://github.com/CelliesProjects/ESP32_VS1053_Stream/compare/dac5ec9...162280b

paulgreg, 2025-01-26:
Strange...
In my `Arduino/libraries/ESP32_VS1053_Stream`, I’ve got that latest commit : 
```
commit dac5ec9 (HEAD -> master, origin/master, origin/HEAD)
Author: Cellie <CelliesProjects@users.noreply.github.com>
Date:   Mon Jan 1 00:30:44 2024 +0100

    Update README.md
```
And my project using `ESP32_VS1053_Stream` compiles nicely 🤔

CelliesProjects, 2025-01-27:
@paulgreg With [ee05f62](https://github.com/CelliesProjects/ESP32_VS1053_Stream/commit/ee05f62444752ac656e0abf13ec3e3d626dcd971) the issue should be fixed.

paulgreg, 2025-01-27:
Indeed ! 
I’ve added my changes (exposing `playChunk` method) over [fix-issue-45](https://github.com/CelliesProjects/ESP32_VS1053_Stream/compare/fix-issue-45) and it worked ! 
Thanks for fixing that 🙏 

By the way, do you think it would be useful to other users to expose `playChunk` method ?

commit ee05f624 (2025-01-27): Fix #45
commit d8d7a0a5 (2025-01-27): Fix #45 (#46)
closed 2025-02-04 by CelliesProjects