CelliesProjects/ESP32_VS1053_Stream #48: FLAC streaming with eps32 nan

Dillmon, 2025-02-22:
I have noticed that FLAC is crossed out in the introduction and that it is not one of the codecs listed in the source code. I believe I have the requirements to stream FLAC from the internet, because the Arduino ESP32 nano has 8MB of PSRAM and has a clock speed of up to 240MHz. 

What would I have to change in the source code to allow me to stream FLAC internet radio?

CelliesProjects, 2025-02-24:
@Dillmon See https://github.com/CelliesProjects/ESP32_VS1053_Stream/issues/8

Dillmon, 2025-02-24:
Thanks. After further research I found out that there is a project that can use the esp32 to stream flac without the need of the VS1053. All that is needed is a dac. https://www.elektroda.com/rtvforum/topic4041603.html

closed 2025-02-24 by Dillmon