CelliesProjects/ESP32_VS1053_Stream #51: VS1053 Low Power Mode for battery operated devices

2dom, 2025-05-20:
I am trying to get the VS1053 into low power mode. Accorting to the data-sheet, pulling XRST low should do the trick. However I still measure around 7mA of power draw. I am using one of those blue boards such as [this](https://de.aliexpress.com/item/1005007137023533.html?src=google&pdp_npi=4%40dis!EUR!3.59!3.19!!!!!%40!12000039533914030!ppc!!!&src=google&albch=shopping&acnt=272-267-0231&isdl=y&slnk=&plac=&mtctp=&albbt=Google_7_shopping&aff_platform=google&aff_short_key=UneMJZVf&gclsrc=aw.ds&&albagn=888888&&ds_e_adid=&ds_e_matchtype=&ds_e_device=c&ds_e_network=x&ds_e_product_group_id=&ds_e_product_id=de1005007137023533&ds_e_product_merchant_id=5367225306&ds_e_product_country=DE&ds_e_product_language=de&ds_e_product_channel=online&ds_e_product_store_id=&ds_url_v=2&albcp=20542169885&albag=&isSmbAutoCall=false&needSmbHouyi=false&gad_source=1&gad_campaignid=19235627935&gbraid=0AAAAAoukdWNKemu_J7Hi0-5ysLct8ocAs&gclid=Cj0KCQjw0LDBBhCnARIsAMpYlApONQHQ6wKvl4VmlW897P1gp4NHsmbp_FraAfdiA7N06EKSE-4p7gEaAuonEALw_wcB) one. Any help would be highly appreciated.

2dom, 2025-05-20:
@CelliesProjects - thanks for coming back to me - really appreciate it.

I checked and you are right - those are VS1003 chips on my blue board. However, according to the datasheet, reset power consumption should also be in the uAmps. Any other ideas?

Sound quality ist ok-ish over here but good to know that there are better boards out there.

CelliesProjects, 2025-05-20:
@2dom I think one of the two LDOs on the board might be the problem. These consume a lot even when idle. If you are serious about a battery powered board you might want to lookup buck converters vs LDOs.

[image]

Chip 1: AMS1117-2.5
Output voltage: 2.5 V fixed.
Quiescent current (Iq): ~5 mA to 6 mA (typical).

Chip 2: AMS1117-3.3
Output voltage: 3.3 V fixed.
Quiescent current: ~5–6 mA (typical).

Even with no load, this is drawn continuously.
Based on your mentioned 7mA probably only one is powered.

2dom, 2025-05-20:
I see - so green board would not help I guess. I now put an N channel MOSFET in the GND supply and that gets the job done apparently. Important to let xRESET float on the microcontroller - the board will get ground via RESET line otherwise (the board has an internal pull up on XRESET anyway so no worries)