CelliesProjects/ESP32_VS1053_Stream #52: Some urls stop immedeately due to 'no remaining bytes'.

Frank-Bemelman, 2025-07-13:
First of all, thanks for this repo. Building a radio globe around it.

I have a database with some 150K urls. Many play without problems. Many do not, for various reasons. 

I extended this library with an extra callback, to get more info as to why a station does not connect (only true/false there) or just quits with a audio_eof_stream callback. I write all these unlucky stations to different text files, to get a nice log of stations to examine later. 

Now I am looking at this particular one, which btw plays perfectly in my PC chrome browser.
https://stream.zeno.fm/8pbaase2w2quv

It connects without problem, even gives me the station name (Harbour Light of the Windwards) but stops immediately with a eof callback.
This happens at the bottom of void ESP32_VS1053_Stream::loop() in ESP32_VS1053_Stream.cpp
Here I added my own extra callback:

```
    if (!_remainingBytes)
    { Serial.println("NO REMAINING BYTES");
      audio_connect_result(FAIL_EOF_NO_REMAINING_BYTES);
        _eofStream();
    }    

```

Nice to know what exit the software took, but I have no clue how to investigate further.

What could be my best next step? Any ideas? Thank you for your time.

PS. I am using a ESP-Wroom-32 with no PSRAM, just the basic thing.

CelliesProjects, 2025-07-14:
Hi @Frank-Bemelman.

On a regular esp32 opening a https stream is very error prone.  
The non https version works for me: http://stream.zeno.fm/8pbaase2w2quv with a quick test on a esp32-s3.

Frank-Bemelman, 2025-07-14:
Ha, thank you - I that's an idea worth trying. I have thousands urls from stream.zeno.fm, all https, and they all show the same behavior. I will try to catch that 'no remaining bytes' and try again with http. And maybe it will fix those other urls I have logged as well. Will let you know how it goes...
Thank you for your time.

Frank-Bemelman, 2025-07-15:
Update!

I added a retry upon https failure, and that is quite a blessing. I have 19999+ stream.zeno.fm urls according to my editor. I also tried to simply convert all https urls to http (saves some 150ms the retry consumes) but also gives http 400 errors on urls that insist on https. But I'll try that anyway, just to collect how many stations have a problem with that, and then perhaps make a retry with the original https url. Just to see what is the better (practical) approach. Ideally, it would be nicer if the ESP32 was less error prone, but what is that is. 

Anyway, I am extremely happy with this result, thank you!

For entertainment purposes only, I stumbled on this one: https://streaming2.toutech.net/jfmweb2
Connects fine on https, also in chrome browser, seems to stream jolly nice, but no sound.

Keep up the good work ;-)

closed 2025-07-15 by Frank-Bemelman